icssw-org/MeshCom-Firmware #33: PlatformIO cannot compile

karamo, 2024-03-23:
**Describe the bug**
PlatformIO cannot compile

```

In file included from src/esp32/esp32_gps.cpp:40:
.pio/libdeps/heltec_wifi_lora_32_V2/EspSoftwareSerial/src/SoftwareSerial.h:24:10: fatal error: circular_queue.h: No such file or directory

.pio/libdeps/heltec_wifi_lora_32_V3/EspSoftwareSerial/src/SoftwareSerial.h:24:10: fatal error: circular_queue.h: No such file or directory

.pio/libdeps/ttgo_tbeam/EspSoftwareSerial/src/SoftwareSerial.h:24:10: fatal error: circular_queue.h: No such file or directory

.pio/libdeps/ttgo-lora32-v21/EspSoftwareSerial/src/SoftwareSerial.h:24:10: fatal error: circular_queue.h: No such file or directory

.pio/libdeps/E22-DevKitC/EspSoftwareSerial/src/SoftwareSerial.h:24:10: fatal error: circular_queue.h: No such file or directory

.pio/libdeps/ttgo_tbeam_SX1268/EspSoftwareSerial/src/SoftwareSerial.h:24:10: fatal error: circular_queue.h: No such file or directory

.pio/libdeps/ttgo_tbeam_AXP2101/EspSoftwareSerial/src/SoftwareSerial.h:24:10: fatal error: circular_queue.h: No such file or directory

Environment             Status    Duration
----------------------  --------  ------------
heltec_wifi_lora_32_V2  FAILED    00:02:18.526
heltec_wifi_lora_32_V3  FAILED    00:01:12.169
ttgo_tbeam              FAILED    00:01:14.727
ttgo-lora32-v21         FAILED    00:01:04.309
E22-DevKitC             FAILED    00:01:01.942
ttgo_tbeam_SX1268       FAILED    00:01:03.550
ttgo_tbeam_AXP2101      FAILED    00:01:03.669
========================================== 7 failed, 0 succeeded
```

karamo, 2024-03-23:
solved with new platformio.ini & readme.md

close issue

oe1kbc, 2024-03-23:
solved with new platformio.ini & readme.md

closed 2024-03-23 by oe1kbc