icssw-org/MeshCom-Firmware #42: T-DECK not supported ?

ok1cdj, 2024-07-12:
Hello, is T-DECK still supported ? 

73 Ondra OK1CDJ

oe1kbc, 2024-07-12:
Hi Ondra,

Yes shure,  and we will work on updates after summer break.

 

73 de Kurt

OE1KBC

 

Ing. Kurt Baumann
Vize-President
ICSSW AMSAT-OE
Höhenstrasse 65
3040 Neulengbach
Austria
T: +43 699 1200 3520
M:  ***@***.***> ***@***.***
H:  <https://icssw.org> https://icssw.org

ok1cdj, 2024-07-12:
Hello Kurt, which ENV is for T-DECK. Firmware is not available in Webflasher.

73 Ondra OK1CDJ

rainerfritz, 2024-07-14:
Hi!

The T-Deck should work now with the online flasher.
Release 4.28b and choose T-Deck.

BR
Rainer

ok1cdj, 2024-07-31:
ok, installed 4.28b without problem..
TNX

ok1cdj, 2024-08-01:
Is there source code for T-DECK available, i think is not part of this FW.

oe1kbc, 2024-08-01:
Hi,No Firmware Source in the Moment but we are working on this to do it in the Future.73 de KurtOE1KBC

closed 2024-10-15 by oe1kbc