icssw-org/MeshCom-Firmware #43: TTGO-LORA32 and battery voltage bug

ok1cdj, 2024-07-31:
On TTGO-LORA32 is not battery voltage displayed properly. Regardless of battery state always 100% is showed. 

According the schematic the IO35 is used for voltage measurement 

https://github.com/LilyGO/TTGO-LORA32/blob/995b37300c8f70b68f01b83ca1ff3ac3e0abc3ec/schematic1in6.pdf

I think vbat_pin is not properly set for TLORA.

rainerfritz, 2024-08-28:
Hi!

You are right. It is set to ADC_UNIT_1 and ADC_CHANNEL_6 which is based on the pinout PIN34, so it should be ADC_CHANNEL_7.

rainerfritz, 2024-09-04:
Please check the current main branch, please compile and flash.

ok1cdj, 2024-09-05:
Hello Rainer, there is some mismatch in GPS library:
`src/loop_functions.cpp:12:10: fatal error: TinyGPSplus.h: No such file or directory
`
was able to compile with `#include "TinyGPS++.h"`

but the voltage measurement still not working, the value shown is 25.7 V.

Regards, Ondra

oe1kbc, 2024-10-10:
solved firmware 4.34c

closed 2024-10-10 by oe1kbc