icssw-org/MeshCom-Firmware #56: Cannot connect AP with SSID starting '#...'

dl7ata, 2024-10-18:
**Describe the bug**
My SSID from accesspoint starts with '#K5...'. I am not able to insert the command  
_--setssid #K5.._ 
via minicom and serial.

**To Reproduce**
Test it yourself by inserting the above command

**Expected behavior**
Configuration param
_—setssid WLANssid_ should accept every possible SSID (maybe as a string, like '#K5...')

**Serial Log**
flash save... 5...
FW V. C4.34c

DK9BT, 2025-01-14:
please re-check that again!

It must be a terminal-issue on you side

I can enter/save a ssid with # without problem

Please try another terminal client

oe1kbc, 2025-01-14:
Please look at 4.34h or higher .. comming soon

oe1kbc, 2025-01-15:
chjecked again - no problem to use

closed 2025-01-15 by oe1kbc