icssw-org/MeshCom-Firmware #57: No more access after typo in filed RF_POWER

dl7ata, 2024-10-19:
After an incorrect entry of  "RF_POWER: 1" (instead of 10), there is no more access to the module.
Neither LAN access nor BLE is possible, oled shows start menu, ending with _...starting now_

**To Reproduce**
Input of _1_ via browser in field "RF_POWER".

**Serial Log**
_[LoRa]...RF_POWER: 1 dBm      Selected output power is invalid for this module!_

Firmware 4.34/c

oe1kbc, 2024-10-19:
Please reflash your node with the WEBFlasher
oe1kfr.com/esptool
But use ERASE before flasing
Your node should be restart agin with default values
73
Kurt

dl7ata, 2024-10-19:
Thanks, that's what I have done. Now its operating again.
Maybe you could modify the WLAN/HTML-access with some restrictions.

DK9BT, 2025-01-14:
@dl7ata 

which hardware are you using?

i can't reproduce the bug in 4.34g

oe1kbc, 2025-01-14:
That's fixed, it was a problem in the flash, after ERASE everything is OK

closed 2025-01-14 by oe1kbc