icssw-org/MeshCom-Firmware #58: Disable "refresh" functionality in HTML-code in message menu

dl7ata, 2024-11-02:
While sending a message via browser, pressing randomly REFRESH, then <enter>  url looks like
_http://192.168.xxx.1x3/refresh/?sendcall=TEST&sendmessage=Diese+Msg+wird+nur+1x+abgeschickt%21_

**To Reproduce**
Push REFRESH button in menu http://192.168.xxx.1x3/message
before sending command

**Expected behavior**
REFRESH should be disabled, otherwise it will send the msg in a loop

DK9BT, 2025-01-14:
when sending a message via the web formular it is send by an url-encoded web-server call

We call for example /message?sendcall=&sendmessage=test

and then the browser is stuck on that site including the url-encodes message.

Probably web-server should return something like a "redirect" back to the /message endpoint (without the url encode)

Then if we would press the browsers "refresh" it would not send the same message again.

@oe1kbc 
May be an additional web-server state that returns 
<meta http-equiv="refresh" content="0; /message">
or sth. like that so that the browser jumps back to the normal non-url-encoded web-page

oe1kbc, 2025-01-14:
Wurde im nightly Build siehe Firmware Chat erledigt 73 de KurtOE1KBC

oe1kbc, 2025-01-14:
Please

oe1kbc, 2025-01-14:
Please check 4.34h it's ready to publish

closed 2025-01-14 by oe1kbc