icssw-org/MeshCom-Firmware #64: T-Deck Plus: Touch failure

ddfeww, 2025-01-09:
**Describe the bug**
Sometimes after startup, the touch-functionality doesn‘t work
„touch: failed“ on boot

**To Reproduce**
putting in external power can solve this sometimes. 

**Additional context**
It seems that TOUCH_INT GPIO(16) is pulled on the t-deck without plus to low, but the T-Deck Plus is pulling it high.

https://github.com/Xinyuan-LilyGO/T-Deck/issues/42

oe1kbc, 2025-01-09:
Sorry this GITHUB is not for T-DECK Hardware

closed 2025-01-09 by oe1kbc

ddfeww, 2025-01-09:
Kurt, auf dem t-deck läuft meshcom. 
In der Firmware ist ein Fehler. 
gruss
Carsten

oe1kbc, 2025-01-09:
Da läuft nur ein Protokoll von MeshCom. Es hat mit der anderen Source nichts zu tun.

Ich habe selber drei Stück und werde mir bei Gelegenheit Zeit nehmen und wieder an dieser Firmware arbeiten.

Ist bei mir nur eine private Leidenschaft welche ich von mal zu mal gerne pflege.

 

btw. Das mit dem Touch-Screen habe ich auch wenn der AKKU schwach ist. Ich kann damit leben.

 

73 de Kurt

OE1KBC

ddfeww, 2025-01-09:
Alles klar, danke Kurt !
Grüsse
Carsten