icssw-org/MeshCom-Firmware #66: Artefacts of bootscreen left on display after complete boot-up of Lilygo T-Beam

dg9vh, 2025-01-09:
**Describe the bug**
Artefacts of bootscreen left on display after complete boot-up of Lilygo T-Beam.

**To Reproduce**
Power down. See that nothing is on display. Power up, wait till completely booted, see artfacts at left side.

**Expected behavior**
sprinkles should be removed from the left border.

**Serial Log**
If possible and needed, will deliver this on demand later...

If no serial Log is available, provide at least the firmware version you are running:
happens on acutal Firmware 4.34e but also happend an release-version before (4.34c)

**Screenshots**
No screenshot, but photo:
[image]

DK9BT, 2025-01-10:
identified the problem and i am working on that now

DK9BT, 2025-01-14:
has been fixed in 4.34g

closed 2025-01-14 by DK9BT