icssw-org/MeshCom-Firmware #67: [Feature Request] Enlarge numbers for GRCs to 5 or 6 digits

dg9vh, 2025-01-09:
Refering to the telegram conversation this is a request to enlarge the width of the GRC-Numbers from now 4 usable digits to 5 or 6 so get more space for smaller groups. With this, a schema like in DMR would be createable with smaller groups for clubs for example.

Just think about it :-)

ddfeww, 2025-01-09:
I would prefer centralized GRC- administration to avoid confusion. 

For the beginning i can do this for you with low budget methods :-)

dg9vh, 2025-01-14:
Well. Depending on the traffic on the groups this could be a point to think about in future. taking an eye on actual traffic, we are good with the 4-digits we have now... I can't have a view into the future but maybe MeshCom get's so much acceptance, that it could be neccessary to divide the traffic... 

But you are right, under actual conditions, and this is, what we can surely talk about, this is not necessary, so I am closing down this FR with an option to reopen it maybe in the future, if conditions changed.

closed 2025-01-14 by dg9vh