icssw-org/MeshCom-Firmware #87: WIFI not working with all Routers

oe1kbc, 2025-01-18:
1 x AP mit 2x SSID (1x XXXXX & 1x "Service") kein MeshWLan
SSID "XXXXXX" ist für den normalen Gebrauch (PC,Handy; Tablet)
SSID "Service" ist für Webcam, SmartHome, IOT, und sonstiges
SSID ist "Hidden SSID"

Config WiFi start

...Webserver off 
...Gateway on 
 ...WIFI-AP off .
..SSID Service .
..PASSWORD Focus-55 
...OWNIP address: 
 ...OWNMS address: 
 ...OWNGW address: 
 ...hasIpAddress: no

folgende Fehlermeldung bei 4.34j

CLIENT STARTED 
============== 
E (41970) wifi:timeout when WiFi un-init, type=4 
-> connecting to BSSID: 00003fcc9ac4 CHAN: 2 
WiFi.power: 80 RSSI:0 Wait WiFI connect ...... 
WiFI ssid<Service> connection error 
E (79471) wifi:timeout when WiFi un-init, type=4 
-> connecting to BSSID: 00003fcc9ab4 CHAN: 2 
WiFi.power: 80 RSSI:0 Wait WiFI connect ...... 
WiFI ssid<Service> connection error 
E (116899) wifi:timeout when WiFi un-init, type=4 
-> connecting to BSSID: 00003fcc68ec CHAN: 2 
WiFi.power: 80 RSSI:0 Wait WiFI connect ...... 
WiFI ssid<Service> connection error 
E (154379) wifi:timeout when WiFi un-init, type=4
 u.s.w.

Dann war ich in meinen Router (Teltonika), und habe im Log gesucht.

Lora-Device antwortet nicht auf positive Anmeldeantwort des Routers.
Habe die "Hidden SSID" herausgenommen.
reboot,
und funktioniert.
Irgendwo zwischen 4.33 und 4.34 hats eine Update der notwendigen Bibliothek gegeben, bzw. der dementsprechende Quelltext in der Firmware wurde erweitert.

Alle Lora-Devices < 4.34 haben das nicht mitbekommen und Funktionieren auch bei Hidden SSID.

Vielleicht kannst das wieder Reparieren und ich werde im Moment das Service offen betreiben, bzw aus 4.33 bleiben.

Danke nochmal für deine Hilfe und sehr gute Arbeit bei diesem schon bald Weltumgreifende Projekt.

LG Fritz

DK9BT, 2025-01-19:
ja das Problem kommt von der Geschichte mit dem Scan wenn man mehrere SSIDs hat....

ich kümmere mich darum und mache einen fix

DK9BT, 2025-01-20:
should be fixed in: https://github.com/icssw-org/MeshCom-Firmware/pull/89

oe1kbc, 2025-01-20:
we prepare next version soon

closed 2025-01-20 by oe1kbc