icssw-org/MeshCom-Firmware #97: KISS-Mode communication via USB

oe1kbc, 2025-01-22:
What is required is a KISS communication via USB interface with APRS compatible protocol

DK9BT, 2025-01-23:
what is your idea behind that? 

Making existing Software like [APRSdroid](https://aprsdroid.org/) compatible with MeshCom?

Or a generall Interface for third party software to access MeshCom?

I ask that cause KISS is basicly encapsulating AX25 Frames and that is what the allready existing Tools expect

check that: https://hackaday.io/project/170710-esp32-tnc-and-audio-relay-for-hfvhf-packet-radio/log/175843-assignable-kiss-interface

oe1kbc, 2025-01-23:
Yes, that's right, an interface with which a lot of existing software can be used together with a LoRa node

ik5xmk, 2025-03-01:
> Yes, that's right, an interface with which a lot of existing software can be used together with a LoRa node

Not only via USB but also via TCP will open a lot possible combinations with existing scenarios

oe1kbc, 2025-03-01:
Hello,
I Think, it’s not the good idea to use TCP because we have Small Devices which are not holding such Stacks Perfect. I have testet some other MESH System and made a decission to use USB only

HelmutEsterer, 2025-03-31:
Hello, 

I recommend the usage of KISS over Bluetooth, as Ricardo Guzman Made!

- It's a really great benefit in my opinion, to have the possibility to use APrsdroid.

With this option it is not necessary to use trackers with GPS. A small simple Heltec V3 would be enough.

With best regards,

Helmut / DJ8MEH

oe1kbc, 2025-04-02:
Hello Helmut,

We intentionally want to use a very dynamically extensible protocol because some people develop their client software using this interface.

73 de Kurt

OE1KBC

oe1kbc, 2026-09-30:
done with v4.35v

closed 2026-09-30 by oe1kbc