icssw-org/MeshCom-Firmware #106: Feature Request - text field to place an individual ALARM code word to initiate a burgular alarm for EMS - HAM interaction

dg1rue, 2025-01-27:
Please cretae a text field to place an individual ALARM code word to initiate a burgular alarm for EMS - Ham interaction.

A Code Word, Pass Phrase or ASCII-"Secret" used in local User Group could support an interaction between EMS and Hams for Emergency circumstances and desasters. If the trigger is placed in a message a Alarm sound different to normal message information would be appreciated.

Thank you for reading and taking care of.

DK5EN, 2026-10-12:
With Remote Management (RM1), you will be able to set PINs in a secure way on a remote node and thus, you will be able to trigger actions. 

Since we have a non encrypted chat (AFu Law), you will need to send a HMAC to authenticate and prove that you are allowed to manage the node. For security reasons, this can't be a simple word, that anyone can place in a chat or a passphrase that a simple replay attack could exploit, once the code word is out in the wild.

DK5EN, 2026-10-25:
Included in https://github.com/icssw-org/MeshCom-Firmware/pull/1238

you can tun on a LED (e.g. white LED on Heltec v3), you can make the LED flash and you have the ability to switch an arbitrary port via RM1 protocol.

closed 2026-10-25 by DK5EN