icssw-org/MeshCom-Firmware #107: Ringbuffer - Problem

karamo, 2025-01-27:
**Describe the bug**
Die Anzeige auf der seriellen Console ist in manchen Fällen zerstückelt.
Ausgabe in den Ringbuffer scheint nicht mit Semaphoren abgesichert zu sein.
**FW 4.34m**

**Serial Log**
```
FLOW raw:4398.55 mV
volt 4398.5 proz 100
:mir scheint der Server tod zu sein, oder?
:mirFLOW raw:4412.73 mV
volt 4412.7 proz 100
scheint der Server tod zu sein, oder?
c_json:{"src_type":"node","type":"msg","src":"OE3WAS-11","dst":"*","msg":"mir scheint der Server tod zu sein, oder?","msg_id":"CC0C40D6"} 130
FLOW raw:4407.41 mV
volt 4407.4 proz 100
```

Hier kommt
```
FLOW raw:4412.73 mV
volt 4412.7 proz 100
```
dazwischen.

oe1kbc, 2025-01-28:
Das hat nichts mit einem Ringbuffer zu tun. Die Eingabe einer Nachricht zur seriellen Console wir von der Putty (oder ähnlich) mit Echo zurück geworfen. Das behindert bzw. bremst die Ausgaben aber keinesfalls.

Lösung:
.) entweder die diversen "DEBUG" auf off einstellen.
.) Zum versenden von Meldungen eine APP verwenden
.) Zum versenden von Meldungen den WEBServer verwenden

73 de Kurt
OE1KBC

closed 2025-01-28 by oe1kbc

karamo, 2025-01-29:
Kann ich nicht so ganz verstehen, denn das hier kommt definitiv von der FW und ist **nicht** das echo vom Terminal:

```
--infoc_json:{"src_type":"udp","type":"msg","src":"OE1XAR-45","dst":"*","msg":"{CET}2025-01-29 10:45:56","msg_id":"6798E194"} 112
--MeshCom C 4.34m
...
```

oe1kbc, 2025-01-29:
Ruf mich einmal an in Worten geht das leichter 73 de KurtOE1KBC