icssw-org/MeshCom-Firmware #108: BME280/BME 680 not working anymore on HELTEC V3 -> FW higher 4.34e

rocktester, 2025-01-29:
...maybe a change in firmware 4.34g and higher ? .... 4.34e works well

SENSORS BME280/680 on PIN 41,42 dont work anymore ... getting I2C ERROR Message if acvtivatet

Maybe something was changed while built in HELTEC Version3.2 ? or in FW 4.34g and following

I2C PINs are also on PCB at test PIN 17 and 18 parallel to display..... but not sure if they can used for the sensors ...but difficult to solder

Regards Mike dg0opk

rocktester, 2025-01-29:
i think this helps .....
testet with 4.34l

#if defined(BOARD_HELTEC_V3)

#define LED_PIN 25

#define RESET_OLED RST_OLED
//#define I2C_SDA 17 // I2C pins for this board
//#define I2C_SCL 18

#define I2C_SDA 41 // I2C pins for this board
#define I2C_SCL 42

regards mik dg0opk

oe1kbc, 2025-02-04:
Fixed at V4.34n

closed 2025-02-04 by oe1kbc