icssw-org/MeshCom-Firmware #137: keine Reaktion auf [User Button] aus der APP

karamo, 2025-02-10:
**Describe the bug**
Szenario:
**TLORA ...-12** normale Funktion, in der APP die [User Buttons] schalten > an der Node im Terminal Aktion sichtbar > Rückmeldung an APP ok.

**TLORA ...-11** keine Reaktion, keine Meldung im Terminal und auch keine in der APP
**TBEAM ...-10** keine Reaktion, keine Meldung im Terminal und auch keine in der APP

Auf allen 3 Nodes **FW v4.34o.02.10** mit Erase & Flash
**MC-APP 4.22 Android**

oe1kbc, 2025-02-11:
Mit Android APP 4.22 pre gerade getestet - Funktion ok

karamo, 2025-02-11:
Ja, für TLORA konnte ich es beheben.
T-Beam hat Rainer schon wahrscheinlich gefunden.

karamo, 2025-02-12:
Aus allen 3 Nodes nach FW **v4.34o.02.12** mit **Erase & Flash** nun behoben.

closed 2025-02-12 by karamo