icssw-org/MeshCom-Firmware #155: Reboot device

DL4QB, 2025-02-12:
Version MeshCom 4.34o (build: Feb 12 2025 / 06:39:41)

Device is rebooting here and there. One time i could log it.
Power was stable, not on Akku.

----------------------
[WIFI]...SSID: BNDU CHAN: 1 RSSI: -21 BSSID: 60:32:B1:99:BF:AB
[WIFI]...connecting to CHAN: 1 BSSID: 60:32:B1:99:BF:AB
[WIFI]...power: 78 RSSI:-21
I[WIFI]...Wait connect 
[WIFI]...connect OK
[WIFI]...now listening at IP 192.168.8.138, UDP port 1990
MPROVGPS: trying 9600 baud <1>
GPS: connected at 9600 baud
GPS serial connected, saved config
 
--setinfo on
setinfo on

21:04:08 NEW-TXT: 032 : xAA00404E H04 S0 T0 M0 DA6MW-3>*:test 13 HW:39 MOD:03 FCS:0645 FW:34:o LH:27
21:04:08Insert own_msg_id:AA00404E
21:04:08 TX-LoRa: 032 : xAA00404E H04 S0 T0 M0 DA6MW-3>*:test 13 HW:39 MOD:03 FCS:0645 FW:34:o LH:27
21:04:19 NEW-TXT: 037 : xAA00404F H04 S0 T0 M0 DA6MW-3>999:test 12345 HW:39 MOD:03 FCS:0762 FW:34:o LH:27
21:04:19Insert own_msg_id:AA00404F
21:04:19 TX-LoRa: 037 : xAA00404F H04 S0 T0 M0 DA6MW-3>999:test 12345 HW:39 MOD:03 FCS:0762 FW:34:o LH:27
21:04:40 NEW-TXT: 035 : xAA004050 H04 S0 T0 M0 DA6MW-3>999:test 789 HW:39 MOD:03 FCS:070C FW:34:o LH:27
21:04:40Insert own_msg_id:AA004050
21:04:40 TX-LoRa: 035 : xAA004050 H04 S0 T0 M0 DA6MW-3>999:test 789 HW:39 MOD:03 FCS:070C FW:34:o LH:27
21:05:12 NEW-POS: 100 ! xAA004051 H02 S0 T0 M0 DA6MW-3>*!5145.50N/00723.21E#/A=000354/P=1006.5/H=43.2/T=19.0/Q=1019.3/R=262;995;888; HW:39 MOD:03 FCS:13A7 FW:34:o LH:27
21:05:12 TX-LoRa: 100 ! xAA004051 H02 S0 T0 M0 DA6MW-3>*!5145.50N/00723.21E#/A=000354/P=1006.5/H=43.2/T=19.0/Q=1019.3/R=262;995;888; HW:39 MOD:03 FCS:13A7 FW:34:o LH:27
21:05:19 TX-LoRa: 032 : xAA00404E H04 S0 T0 M0 DA6MW-3>*:test 13 HW:39 MOD:03 FCS:0645 FW:34:o LH:27
21:05:28 TX-LoRa: 037 : xAA00404F H04 S0 T0 M0 DA6MW-3>999:test 12345 HW:39 MOD:03 FCS:0762 FW:34:o LH:27
21:05:45 NEW-TXT: 033 : xAA004052 H04 S0 T0 M0 DA6MW-3>*:test 345 HW:39 MOD:03 FCS:0681 FW:34:o LH:27
21:05:45Insert own_msg_id:AA004052
21:05:45 TX-LoRa: 033 : xAA004052 H04 S0 T0 M0 DA6MW-3>*:test 345 HW:39 MOD:03 FCS:0681 FW:34:o LH:27
21:05:53 TX-LoRa: 035 : xAA004050 H04 S0 T0 M0 DA6MW-3>999:test 789 HW:39 MOD:03 FCS:070C FW:34:o LH:27
TimeClient now (UTC): 20:06:42

abort() was called at PC 0x40085be1 on core 0

Backtrace: 0x40084571:0x3ffe2900 0x40096dcd:0x3ffe2920 0x4009c509:0x3ffe2940 0x40085be1:0x3ffe29c0 0x40085c0e:0x3ffe29e0 0x40085d85:0x3ffe2a10 0x4012eb62:0x3ffe2a30 0x4012dddf:0x3ffe2a50 0x4012d5d5:0x3ffe2a70 0x400935a5:0x3ffe2a90 0x40091db2:0x3ffe2ab0 0x40091e3c:0x3ffe2ad0 0x40092329:0x3ffe2b00 0x401be4db:0x3ffe2b30 0x401b7453:0x3ffe2b50 0x401b75d1:0x3ffe2e60 0x401cbcd9:0x3ffe2e90 0x401557c6:0x3ffe2ec0 0x40155863:0x3ffe2ef0 0x40150fce:0x3ffe2f70 0x40154509:0x3ffe2fb0 0x401409c5:0x3ffe2ff0 0x40143d6f:0x3ffe3040 0x40147e02:0x3ffe3070 0x40138b4d:0x3ffe3090

ELF file SHA256: fca17079369ee915

E (1755) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
E (590) esp_core_dump_flash: No core dum   ѥѥ   found!
E (590) esp_core_dump_flash: No core dump partition found!
Terminal disconnected: BreakError: Break received

DL4QB, 2025-02-12:
Device was flashed with Erase before, so clean install

oe1kbc, 2025-02-12:
From where does this new messages came from?

DL4QB, 2025-02-13:
Device or what you mean about?   This ist the E22. Message was sent via Webinterface

oe1kbc, 2025-02-13:
TKS, Was a console active on the serial interface? I have some issues from this direction

DL4QB, 2025-02-13:
yes, it was connected.

oe1kbc, 2025-02-24:
We keep getting reports that unwanted hardware resets often only occur when the hardware is operated via a serial console on the USB interface. Can you operate this node without a console and determine whether this error persists? Thanks

oe1kbc, 2025-02-27:
fixed - error on command functions

closed 2025-02-27 by oe1kbc