icssw-org/MeshCom-Firmware #156: Exit from SafeBoot via RST or Timeout

karamo, 2025-02-12:
Ohne einem Web-Zugriff bibt es keine Möglichkeit, den SafeBoot zu beenden.

<ins>Mögliche Lösungen:</ins>
* Timeout: Nach 3-5 Minuten macht die Node selbsttätig ein REBOOT in den vorigen Normalzustand.
* ein REBOOT via RST (Taste) = HW RESET bewirkt ein Ende des SafeBoot-Modus.

oe1kbc, 2025-02-24:
Is it still upright? Who can test it again?

karamo, 2025-02-24:
**FW 4.34p** TLORA
❌Nach RESET (RST-Taste) immer noch im Safeboot-Modus.
❌Reagiert auch auf keine Befehler via Terminal.
❌nach ≈7 Min. auch kein Timeout und selbständige Rückkehr in den Normal-Modus.
❌trennen der Versorgung - bootet wieder in den Safeboot-Modus.

Es gesteht ausschließlich die Möglichkeit, den Safeboot-Modus zu benenden, wenn man über den Web-Browser zugreift.

karamo, 2025-03-05:
**FW 4.34p03.04**
✅ selbständiger RESTART in den Normalmodus nach 3 Minuten.

Zumindest eine Möglichkeit existiert jetzt.
Falls man sich damit zufrieden geben kann, kann dieses Issue geschlossen werden.

oe1kbc, 2025-03-14:
The display now also shows the IP address and the GW address during the OTA process if in Wifi mode if in WfiFi AP mode also the IP address

FW >= v4.34t

closed 2025-03-14 by oe1kbc