icssw-org/MeshCom-Firmware #172: mheard wrong date

DL4QB, 2025-02-22:
The Node was running the wohle day and seems ok.

After a check later or mheard, i got this date.  GPS is on and has a fix with 7 Sats.
So time and date should be ok, but seems not so as we can see here.

[image]

[image]

oe1kbc, 2025-02-24:
Have you noticed an incorrect date only on the ESP32/E22? Or does this also occur on other hardware?
I cannot reproduce it on my countless nodes at the moment and therefore cannot determine the actual reason yet.

oe1kbc, 2025-02-27:
fixed with Version v4.34p02.27

closed 2025-02-27 by oe1kbc