icssw-org/MeshCom-Firmware #173: [Feature] Shut-Down-Command

DL9CL, 2025-02-23:
For people who use T-Beam with integrated battery or using some exposed nodes without direct fast access (repeaters), they need a possibility to shut-down their node by remote-command like "--poweroff".

Can you please consider this issue for further releases ?

oe1kbc, 2025-02-24:
These or similar requests have already been made. I myself operate 5 locations with colleagues that are up to 150 km apart. The most important thing for me is that these locations can be programmed remotely without personal access and that they do not get stuck in a mode that can only be resolved by personal access. I think a remote shutdown is very dangerous if you do not develop a stable wake-up process. I have not yet found a good option for this.

ik5xmk, 2025-03-01:
> For people who use T-Beam with integrated battery or using some exposed nodes without direct fast access (repeaters), they need a possibility to shut-down their node by remote-command like "--poweroff".
> 
> Can you please consider this issue for further releases ?

For remote control on and off (I think of possible hardware malfunctions) I have inserted a controlled electrical socket on all my systems, manageable remotely. In any case, wifi must be present if the system must be able to be managed remotely. Otherwise, a command (encoded) could be thought of that can be sent via RF as text in the message, which is not propagated/visible to the server except as a receipt ACK on the recipient node, to assist the sender...

closed 2025-03-18 by oe1kbc