icssw-org/MeshCom-Firmware #174: [Feature] Option for Range-Test (Reichweitentest)

DL9CL, 2025-02-23:
For some cases it would be helpful to send more often periodic signals via hf than now (like 1 signal per minute), enriched with a counter.
This is implemented in meshtastic and it is a very useful feature. when this function is activated, the node sends a number, increased every minute for a defined period (e.g. max 1hour).

This must be for hf only and not for meshing. (this could be a serious issue to develop :-) )

oe1kbc, 2025-02-24:
Thanks for passing on the idea. I will include the topic in the development of the new protocol (5.0). As you write, the MESH is to be completely excluded - it only makes sense then - and it would certainly be disruptive in ongoing operation as a backpack in protocol 4.0.

oe1kbc, 2026-09-30:
use ping/pong version v4.35v

closed 2026-09-30 by oe1kbc