icssw-org/MeshCom-Firmware #207: Bug: If no wifi is available, Node display hung in "Starting now"

DL4QB, 2025-03-12:
4.34r and E22,

if you boot the node without wifi available, the node stucks in display like attached:

Also it shows no IP in Webinterface i.e.

If you having back the wifi, it still says no IP and still in "starting now"

[image]

[image]

DL4QB, 2025-03-12:
t-beam the same

oe1kbc, 2025-03-13:
This is a planned process; we wait approximately 39 seconds until the node detects that no Wi-Fi is available. Then the node starts normal operation. If no Wi-Fi is found, it will search for Wi-Fi every 15 minutes to avoid constantly burdening the node. Only when a stable Wi-Fi connection is reestablished will all parameters be displayed again.

It would not be correct to proceed otherwise, as this situation is an exception.

closed 2025-03-13 by oe1kbc

oe1kbc, 2025-03-13:
The start screen is only replaced with the first message that comes in via LoRa-HF

oe1kbc, 2025-03-13:
in the next version we include a startscreen with Wifi down please wait or something like that