icssw-org/MeshCom-Firmware #210: Bug? / Help: GPS does not work anymore

DL4QB, 2025-03-13:
E22 Board , 4.34R, GPS worked nice in the past. I´m not sure if this has something to do with the Firmware, i don´t check the count of SATs every day. So it could be, that this situation is longer time on-going. 

However, Sat in Webinterface stays to 0. GPS Modul is blinking, means usually it received one or more SATs.

My log with gpsdebug on. There is not that much incoming after 10 minutes. 

So please verify this with E22 Board. If that works on your side, than maybe my GPS module is broken.

============
CLIENT SETUP
============
[INIT]...initMheard
[INIT]...init_flash
[INIT]...init_batt
[INIT]...setupBMX280 - I2C:76
[INIT]...[INIT]...BME280 startet
[INIT]...MCP23017 not found
[INIT]...INA226 not found
[INIT]...RTC not found
[INIT]...ONEWIRE - GPIO:25
[INIT]..._GW_ID: BF689964
[INIT]...Initialize clock: ok
[INIT]...Auto detecting display:
[  1848][E][Wire.cpp:499] requestFrom(): i2cWriteReadNonStop returned Error -1
[INIT]...Display type: 3F
[INIT]...OLED Display is SH1106
[INIT]...LoRa Modem Initializing ... LoRa-Chip success!
[LoRa]...RX_BOOSTED_GAIN: 0
[LoRa]...RF_FREQUENCY: 433.175 MHz
[LoRa]...RF_BANDWIDTH: 250 kHz
[LoRa]...RF_SF: 11
[LoRa]...RF_CR: 4/6
[LoRa]...RF_POWER: 3 dBm
[LoRa]...PREAMBLE: 8 symbols
[SX126x/SX1262X] Starting to listen ... success!
[SX12xx] All settings successfully changed!
[INIT]...BLE-Device started with BLE-Name <MC-9964-DL4QB-3>
[INIT]...BLE Devicename <88:13:bf:68:99:66>
[INIT]...Waiting a client connection to notify...
==============
CLIENT STARTED
==============
[WIFI]...SSID: BNDU CHAN: 1 RSSI: -44 BSSID: 60:32:B1:99:BF:AB
[WIFI]...connecting to CHAN: 1 BSSID: 60:32:B1:99:BF:AB
[WIFI]...power: 78 RSSI:-45
-[WIFI]...Wait connect
[WIFI]...connect OK
[WIFI]...now listening at IP 192.168.8.122, UDP port 1990
-gpsdeGPS: trying 9600 baud <1>
GPS: connected at 9600 baud
GPS serial connected, saved config
-----------check GPS-----------
▒bnewData:0 SAT:0 Fix:0 UPD:0 VAL:0 HDOP:0
PuTTY-----------check GPS-----------
        @newData:0 SAT:0 Fix:0 UPD:0 VAL:0 HDOP:0

oe1kbc, 2025-03-14:
Hi,
I need a serial console printout with --gpsdebug enabled. Only that will give us any information about why and how GPS is running.

DL4QB, 2025-03-14:
Hi Kurt, it is already attached look at gps.txt

oe1kbc, 2025-03-14:
Sorry with --gpsdebug on please

oe1kbc, 2025-03-14:
Closed because of no FW issue

closed 2025-03-14 by oe1kbc