icssw-org/MeshCom-Firmware #213: BUG: Messages will not be gated

DL4QB, 2025-03-14:
Tnx for the phone call regarding not gated mesages if Wifi fails. 

So my partner around here (with GW) have already version R running now. You mentioned, that q is minimum, my GW Partners have now R and myself version S.

So i did the Test again:

- My Node DL4QB-3 is running with GW on and Mesh on and Wifi available
- I send a message via T-Deck Plus 4.34q ( Call DA6MW)
- Message will be delivered of course and i can see that at the dashboard. 

NOW:

I switch off Wifi (power outage simulation)

- I send again a message via T-Deck (DA6MW)
- Message is received by the Node DL4QB-3
- In the serial log you can see DL4QB-3 has send out the message via HF:

08:59:49 TX-LoRa: 039 : x8A0EC0BA H03 S1 T0 M01 DA6MW-2,DL4QB-3>*:test 2 HW:46 MOD:8/8 FCS:0A0A FW:30:q LH:A7

But none of this message will gated by any other GW around and near me, running Version r already. 

You also mentioned, if i boot the node (wifi still not available) that works again. But it doesen´t.

**So, please, you have the same devices like me, try to do it by yourself.**

I attached the Log gw.txt  here. Look for DL4QB-3, DA6MW and the messages test, test1, test2, test3

**I just want to help (and not bother!!) to fix a problem,if a poweroutage fails and GW Stations without Wifi will not gated by other stations. It is no problem if the analyse takes some time, we don´t need a solution ad hoc :-)**

oe1kbc, 2025-03-14:
Thanks for the details, bugs found and fixed. I'll release v434s.03.15 this evening.

closed 2025-03-14 by oe1kbc

DL4QB, 2025-03-15:
Great!!! :-)  But i just can not find v434s.03.15

oe1kbc, 2025-03-15:
It is 4.34t 73 de KurtOE1KBC