icssw-org/MeshCom-Firmware #216: [Feature] more intelligent time signal handling

DK5EN, 2025-03-14:
Currently, every GW sends out the time signal. So far, so good.

But we also have time signals that get meshed and this blocks quite some air time.

12:44:15:67D39903 2 101 8/8 LH:AB OE1XAR-45,DK5EN-12>* :{CET}2025-03-14 11:44:14
12:44:21:67D39903 1 101 8/8 LH:AB OE1XAR-45,DK5EN-12,DL2JA-1>* :{CET}2025-03-14 11:44:14
12:44:26:67D39903 0 101 8/8 LH:2B OE1XAR-45,DK5EN-12,DL2JA-1,DB0ED-1>* :{CET}2025-03-14 11:44:14

DL2JA-1 is a gateway, it doesn't need to mesh my time signal vom DK5EN-12
DB0ED-1 is only a mesh, thus it makes sense to send out the time signal

Simply stop digipeating time signals, if you are a source of a good time signal, as you are a gateway.

DK5EN, 2025-03-14:
IMHO: like APRS hops, time sources have a stratum level. With every hop, the stratum level drops.

oe1kbc, 2025-03-14:
Can you explain exactly how you see this happening in a MESH network? Ultimately, all nodes in an RF cloud need to receive the time signal. Whether one or more gateways are involved in an RF cloud is unknown. To prevent this from spreading arbitrarily, each timestamp has an MSG ID, and if a node has passed on this MSG ID and hears it again, nothing happens.

oe1kbc, 2025-03-14:
No more answer so i close this not relevant issue

closed 2025-03-14 by oe1kbc