icssw-org/MeshCom-Firmware #217: [Bug Report] Messages no longer received

DK5EN, 2025-03-14:
**Describe the bug**
On 14th March, all the group chat no longer has been received.
Gateway is turned on at -99
DK5EN-3 is listening

I am not able to send Personal Messages to DK5EN-99 and DK5EN-3 from Telegram Bot

After reboot my -99 gateway can receive messages. But it doen't route messages to DK5EN-3

**To Reproduce**
Steps to reproduce the behavior. Please describe as clear as possible:

Difficult, but I am not the only one that now rebooted. Maybe something on the server backend happened?

**Expected behavior**
A clear and concise description of what you expected to happen.

I want to send a message from Telegram Bot to DK5EN-3
But DK5EN-3 never receives the message, as my GW -99 never sends this message via HF.

**Serial Log**
If possible provide a serial log of the command --info as well if needed the output of the debug.
You can find the commands for enabling debugging on this page:
https://icssw.org/en/meshcom-kommandos-cl-gw/

not possible

If no serial Log is available, provide at least the firmware version you are running:

**Screenshots**
If applicable, add screenshots to help explain your problem.

See below .. 

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - App Version [e.g. 22]

Smartphone details not necessary 

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

**Additional context**
Add any other context about the problem here.

DK5EN, 2025-03-14:
After sending a message from my -3 (via -99), the server in Vienna knows again where to send messages.

[image]

[image]

[image]

[image]

oe1kbc, 2025-03-14:
No possible form of communicfating Infomation ... sorry

closed 2025-03-14 by oe1kbc