icssw-org/MeshCom-Firmware #224: [Feature] store messages in Mesh for later retrieval

pweichsel, 2025-03-16:
Can we think about how to establish some message storage within the "mesh" until a node comes back "online"? Use case would be distributing information for nodes or even groups although they are temporarily offline, such as out of reach, no power or i.e. in emcomm/off grid situations, running on a lean power budget. An implementation could work similar to JS8Call, where one "online" node A is advised by a specific message format to save a message for later retrieval by another node B. As node B comes back online, it ask for "any message for me" and then gets a message delivered by node A - this would likely even work across the mesh, provided only that the retrieval request from node B reaches node A and the answer is delivered vice versa.

DK5EN, 2026-10-02:
S&F now works over LoRa for all Nodes that you can hear directly in Version v4.40a

DK5EN, 2026-10-05:
As the feature is now included in v4.40a for the LoRa side, we are closing this request. 

There is another issue now open, that covers the internet side of things.
https://github.com/icssw-org/MeshCom-Firmware/issues/1188

closed 2026-10-05 by DK5EN