icssw-org/MeshCom-Firmware #227: [BUG] if no Display false detection

karamo, 2025-03-16:
**FW V4.34u**
**T-Beam no OLED**

Wenn an dem T-Beam kein Display angeschlossen ist, wird dennoch eines fälschlicherweise erkannt.

Reboot
```
[INIT]...Auto detecting display:
[  2194][E][Wire.cpp:499] requestFrom(): i2cWriteReadNonStop returned Error -1
[INIT]...Display type: 3F
[INIT]...OLED Display is SH1106
```

`--info`
```
--MeshCom 4.34u (build: Mar 16 2025 / 12:33:30)
...UPDATE: 2025-03-16 18:12:39
...Call: <OE3WAS-13> ...ID 8488A210 ...NODE 45 ...UTC-OFF 1.000000
...DISPLAY on
```

`--showi2c`
```
[I2C] ... Scanner started
showi2c --[I2C] ... Scanner
[I2C] ... device found at address 0x34 AXP192/2101
```

karamo, 2025-03-17:
Das Problem ist in **FW 4.34v** für `Reboot` gelöst.

Aber es wird bei `--info` unpassend **DISPLAY ON** angezeigt und auch an BLE (APP) übermittelt. Das ist irritierend.

oe1kbc, 2025-03-18:
solved on next version

closed 2025-03-18 by oe1kbc