icssw-org/MeshCom-Firmware #228: [BUG] Trying to get Hotspot maybe too short

DL4QB, 2025-03-16:
I´m using a raspberry configured as a Hotspot. 

Thats works really nice! 

**Any other device can connect.** Smartphone Android and also Apple.
I can connect with my PC / Win11 as well and also with anohter Linux.

So Hotspot ist working fine. 

The Node with now 4.34U can´t get a connect.

Attached the Lofile with --setinfo on

Also after more 5 minutes, same results. Maybe the waiting time to get an IP is too short?
I see on my smartphone it takes a moment longer to connect, but it is successful.

[wifi.txt](https://github.com/user-attachments/files/19274483/wifi.txt)

oe1kbc, 2025-03-16:
How long should we wait? Please state seconds.

Tks

73

Kurt

 

Ing. Kurt Baumann
Vize-President
ICSSW AMSAT-OE
Höhenstrasse 65
3040 Neulengbach
Austria
T: +43 699 1200 3520
M:  ***@***.***> ***@***.***
H:  <https://icssw.org> https://icssw.org

oe1kbc, 2025-03-18:
Solved on next version

closed 2025-03-18 by oe1kbc