icssw-org/MeshCom-Firmware #237: Integration Heltec Wireless Tracker 433 MHz Version

Rhodeia08, 2025-03-23:
Hallo 

I am try to install my heltec wireless tracker 433 mhz with the firmware 4.34v I use the heltec V3.0. It run but the display not show anything it will dark.

I can see my node inside my home network. I can receive and send msg. The gps inside will be run too. Aprs will work without any problem. Perhaps it's possible to adapt the actual firmware to this wireless tracker

DG2MCM

WOLFGANG

elFallino, 2025-04-21:
The Heltec Wireless Tracker (HWT) uses a different display (160x80px TFT connected via eSPI as far as I could find out).

The Heltec V3 is using a 128x64 OLED connected via i2c.

Supporting the HWT would be technically possible (at least I think so) but would need some time (and devices to test code on).

mi-gri, 2025-04-23:
Hi! I was able to successfully flash the Heltec Wireless Tracker as Heltec E290 (Heltec v3 did not work for me!). Apparently everything works except for the display.

closed 2025-07-14 by oe1kbc