icssw-org/MeshCom-Firmware #240: [BUG] mysterious Message on BLE

karamo, 2025-03-24:
FW **v4-34v.03.23** auf TLORA

Wenn man via BLE verbunden ist und am Terminal `--help` eingibt, oder via APP absendet, dann kommt der letzte Teil der INFO davon auf BLE als Message:

```
--help
commandAction [--help] ble:1
<3A>BLEtoPhone RingBuff added len=175 to element: 12
B3 3A 67 EB 12 01 34 72 65 73 70 6F 6E 73 65 3E 3A 2D 2D ... 7E 67 E0 4D 50 
MeshCom 4.34v commands
--setcall  set callsign (OE0XXX-1)
--setname  set first name
...
--webpwd    xxxx
--mesh    on/off
--softser on/off/send/app/baud/fixpegel/fixtemp
--spectrum   run spectral scan

[LOOP] WRITE BLE
toPhoneWrite:13 toPhoneRead:13 buff:response>:--MeshCom 4.34v commands
--info show info
--reboot  Node reboot
--pos show lat/lon/alt/time info
--sendpos send pos now
--sendtrack send LORAAprs now
```

oe1kbc, 2025-03-24:
Das habe ich absichtlich so reduziert.73 de KurtOE1KBC

oe1kbc, 2025-03-24:
This is intentionally reduced in the FW

closed 2025-03-24 by oe1kbc