icssw-org/MeshCom-Firmware #254: [Feature] Statt UTC-Offset den TZ-String verwenden

karamo, 2025-03-30:
Der **TZ-String** (**Time-Zone-String**) beschreibt die Umstellung auf Sommerzeit:
z.B. für **Zeitzone Wien/Berlin**: `CET-1CEST,M3.5.0/2,M10.5.0/3`
_"Die Uhren werden jedes Jahr am letzten Sonntag im März (3) um 02.00 Uhr (2) eine Stunde vorgestellt und am letzten Sonntag im Oktober (10) um 03.00 Uhr (3) eine Stunde zurückgestellt (Ortszeit DACH)."_
Dieser "letzten Sonntag" wird durch die 5 dargestellt.

siehe: https://www.di-mgt.com.au/wclock/help/wclo_tzexplain.html

Damit wäre die automatische Umstellung auf Sommerzeit möglich.

oe1kbc, 2025-03-30:
Sorry wir haben viele MeshCom Freunde auch ausserhalb von Europa, da muss uns was besseres einfallen

closed 2025-03-30 by oe1kbc

oe1kbc, 2025-03-31:
>   Ich möchte das doch nochmals aufgreifen. Ich bin offenbar nicht ganz verstanden worden.
> Wenn man die Möglichkeit vorsieht, so einen TZ-String als Parameter zu setzen, dann kann ja jeder einen eigenen TZ-String entsprechend seiner Zeitzone eintragen.
> Das würde der FW ermöglichen, die Zeit korrekt einzustellen, indem dieser (individuelle) TZ-String in der FW ausgewertet wird.
>
> Ich möchte das doch nochmals aufgreifen. Ich bin offenbar nicht ganz verstanden worden.
> Wenn man die Möglichkeit vorsieht, so einen TZ-String als Parameter zu setzen, dann kann ja jeder einen eigenen TZ-String entsprechend seiner Zeitzone eintragen.
> Das würde der FW ermöglichen, die Zeit korrekt einzustellen, indem dieser (individuelle) TZ-String in der FW ausgewertet wird.
> PS: Bei Meshtastic funktioniert das einwandfrei.

Aber was ist der Unterschied zu einem Offset? Die UTC Zeit muss ich ja auch beziehen wenn kein NTP oder GPS zur Verfügung steht73 de KurtOE1KBC

karamo, 2025-03-31:
> Aber was ist der Unterschied zu einem Offset?

Ich gebe 1x den TZ-String ein und es kann dei FW dann automatisch aus der UTC auf die LT umrechnen.
Man muss nicht jedes Mal bei Sommerzeit-Wechsel den UTC-Offset ändern, da im TZ-String die genauen Daten, das Datum der Umstelung kodiert ist.

oe1kbc, 2025-03-31:
Ok verstehe 73 de KurtOE1KBC