icssw-org/MeshCom-Firmware #266: [Feature] MCP23017 --io status on BLE

karamo, 2025-04-05:
**FW 4.34w**

Zur Steuerung des **MCP23017** sind bereits folgende Befehle verfügbar:
* `--setio clear`
* `--setio { A0..A7,B0..B7 } { in | out }`
* `--setout { A0..A7,B0..B7 } { on | off }`
* `--io`

All diese Befehle funktionieren bereits über BLE, aber es fehlt die Rückmeldung des Gesamtzustandes, der mittels `--io` derzeit nur am Terminal angezeigt wird. Um eine vollständige Steuerung des **MCP23017** via BLE zu ermöglichen, ist diese Rückmeldung des Gesamtzustandes erforderlich.  
Die Frage zur Sicherheit ist durch den BLE-Pin gegeben, der ja individuell einstellbar ist.

Ich habe folgende Erweiterung in `command_functions.cpp` eingebaut und mit meiner eigenen APP getestet:
```
    if(bIO)
    {
        if(ble)
        {
            // OE3WAS addition
            uint16_t t_io = meshcom_settings.node_mcp17io;
            uint16_t t_out = meshcom_settings.node_mcp17out;
            uint16_t t_in = meshcom_settings.node_mcp17in;

            // reset print buffer
            memset(print_buff, 0, sizeof(print_buff));
            String iooutA="";
            String iovalA="";
            String iooutB="";
            String iovalB="";
            char temp[1];

            iooutA = ""; iovalA = ""; iooutB = ""; iovalB = "";
            for(int io=0;io<16;io++)
            {
                bool bOut = ((t_io & 0x0001) == 0x0001);
                bool bOutValue =((t_out & 0x0001) == 0x0001);
                bool bInValue = ((t_in & 0x0001) == 0x0001);

                if(io < 8) {
                    sprintf(temp,"%s",(bOut ? "1":"0"));
                    iooutA = iooutA + temp;
                    sprintf(temp,"%s",(bOut ? (bOutValue ? "1":"0"):(bInValue ? "1":"0")));
                    iovalA = iovalA + temp;
                } else {
                    sprintf(temp,"%s",(bOut ? "1":"0"));
                    iooutB = iooutB + temp;
                    sprintf(temp,"%s",(bOut ? (bOutValue ? "1":"0"):(bInValue ? "1":"0")));
                    iovalB = iovalB + temp;
                }
                t_io = t_io >> 1;
                t_out = t_out >> 1;
                t_in = t_in >> 1;
            }

            JsonDocument iodoc;
            iodoc["TYP"] = "IO";
            iodoc["MCP23017"] = bMCP23017;
            iodoc["AxOUT"] = iooutA;
            iodoc["AxVAL"] = iovalA;
            iodoc["BxOUT"] = iooutB;
            iodoc["BxVAL"] = iovalB;
            serializeJson(iodoc, print_buff, measureJson(iodoc));

            // clear buffer
            memset(msg_buffer, 0, MAX_MSG_LEN_PHONE);

            // set data message flag and tx ble
            msg_buffer[0] = 0x44;
            memcpy(msg_buffer +1, print_buff, strlen(print_buff));
            addBLEComToOutBuffer(msg_buffer, strlen(print_buff) + 1);
        }
...

```
Die Ausgabe auf BLE stellt einen Kompromiss zwischen guter menschlicher Lesbarkeit und Länge des Strings dar:  
z.B.:  
```D{"TYP":"IO","MCP23017":true,"AxOUT":"10000000","AxVAL":"01111111","BxOUT":"00000000","BxVAL":"11111111"}```

Bitte einbauen.

karamo, 2025-04-06:
Achtung: muss auch die terminierende 0 enthalten!
**`char temp[2];`**

karamo, 2025-04-07:
Achtung: verbesserer Code!
Es geht natürlich noch einfacher, kürzer und schneller!

oe1kbc, 2025-04-07:
Code ist in v4.34w.04.07 includiert

closed 2025-04-07 by oe1kbc