icssw-org/MeshCom-Firmware #296: [Bug/Feature] Missing BLE for ANALOG

karamo, 2025-04-16:
**FW 4.34x**

Mit der Erweitunge der ANALOG-Parameter, um eine Spannung an einem ADC-GPIO messen zu können ist nur bedingt brauchbar, da ...

• die gemessenen Werte nur im Terminal als DEBUG-Ausgabe angezeigt werden
• und im Web-Client

Das Setzen der Parameter erfolgt mit:  
```
--analog GPIO 35  
--analog FACTOR 1.0  
--analogcheck {on|off}
```

Die gesetzten Parameter **"ANACHK":true,"ANAFAC":1,"ANAPIN":35** bekommt man über den Befehl `--nodeset` auch auf BLE, aber nicht den Messwert.

**Bitte eine Ausgabe des ANALOG-Messwertes nach BLE ergänzen.**

karamo, 2025-04-17:
z.B. den BLE-JSON von `--wx `ergänzen mit: **"ADC":xxxx.xx**

karamo, 2025-04-17:
den BLE-JSON von `--wx` ergänzen mit: **"ADC":xxxx.xx**
```
            wdoc["VPOW"] = meshcom_settings.node_vpower;
            wdoc["ADC"] = fAnalogValue; // OE3WAS: Wert verfügbar ?

```

oe1kbc, 2025-04-20:
fixed with version v4.34x.04.20

closed 2025-04-20 by oe1kbc