icssw-org/MeshCom-Firmware #309: [Protocol definition] GPS altitude in feet instead of meters

fmoessbauer, 2025-04-20:
**Describe the bug**
The Readme file states the following regarding the GPS altitude parameter:

> /A=HHHH GPS sea level (m) int 0 - 9999

However what is transmitted via AX25 is a measurement in feet. Further, the format is different (as seen in this raw trace from LoRa):

```
/B=100/A=001631
         ^ -- 6 digits instead of 4
```

The specified range 0-9999 further does not match the definition on ICSSW, which states 0-40000 (also in m).
https://icssw.org/en/meshcom-2-0-protokoll/

When changing this, please also consider replacing GPS with GNSS, as likely other GNSS systems can be used as well.

**Expected behavior**
The implementation should match the protocol definition

73 - de DL1FMP

oe1kbc, 2025-04-20:
Fixed on project-page

closed 2025-04-20 by oe1kbc

fmoessbauer, 2025-04-20:
Thanks for the quick fix. Please also fix it in the Readme.
Further, please clarify if the max value is 400'000 as on the project page, or 40'000 as in the Readme.

oe1kbc, 2025-04-22:
Done