icssw-org/MeshCom-Firmware #334: [BUG] E22-DevKitC-1-N16R8 - corr definitions

karamo, 2025-04-27:
**FW 4.34x.04.24**

Konsolidierte & bereinigte `platformio.ini` & `configuration.h`

In **`esp32_main.cpp`** daher wegen konsistenter Benennung **defined(BOARD_E22_S3)** ergänzt und **RXEN** & **TXEN** umbenannt (folglich müssen für die **variants E22-DevKitC** u.a. auch noch diese Umbenennung erfolgen).

[image]

[E22-DevKitC-1-N16R8_variants.zip](https://github.com/user-attachments/files/19928899/E22-DevKitC-1-N16R8_variants.zip)

oe1kbc, 2025-04-27:
fixed > v4.24x.04.26

closed 2025-04-27 by oe1kbc

karamo, 2025-04-27:
!!! TYPO fixed !!!
in configuration.h
#define MODUL_HARDWARE **EBYTE_E22**

[E22-DevKitC-1-N16R8_variants.zip](https://github.com/user-attachments/files/19929000/E22-DevKitC-1-N16R8_variants.zip)