icssw-org/MeshCom-Firmware #354: Heltec V2.1 - no OLED found

DL5SMB, 2025-05-01:
Hello everyone,

I recently reflashed my Heltec V2.1 from Lora-Aprs to Meshcom. 

Everything seems to work except the OLED, which is not being found. If I try to scan the I2C bus,
I just get the message "No devices found". But as it works with Lora-Aprs, the hardware is abviously okay.

Enclosed there is the Log-file, hopefully its helps understanding the problem.
[log.txt](https://github.com/user-attachments/files/20004874/log.txt)

rainerfritz, 2025-05-05:
Hi! 
Can you find the V2 label printed next to the LoRa antenna connector? If not you have the V1 version which is not officially supported. Are you using any additional sensors connected to the I2C bus?

BR
Rainer

karamo, 2025-05-09:
Das ist definitiv ein altes V1 Board

DL5SMB, 2025-05-20:
I had another look at my boards this week - it really does show V2.

[image]

However after I reuploaded the latest LORA-APRS firmware again, the display would not work anymore either.

I had to pull a version from 2023 (which did noct compile with the latest platformio-arduinoespressif32 ( version 6.81?),
so I had to reinstall an older version - I think 6.32.  And once that was done the OLED worked again.
So I am not really sure, but maybe its problem with the development kit.

closed 2025-05-29 by oe1kbc