icssw-org/MeshCom-Firmware #357: [Feature] Rufzeichen als anklickbare Links zu aprs.fi

aendes, 2025-05-03:
Hallo,

ich fände es klasse, wenn in wer WebUI in den mheard/path/messages die empfangen Rufzeichen Links zu aprs.fi wären. Es könnte dann über einen einfachen klick ein neuer Browser- Tab aufgehen, in dem auf aprs.fi der Standort angezeigt wird.

Dies wäre möglich, indem z.B. das Rufzeichen in der Tabelle als Link dargestellt wird:

<a target="_blank" href="https://aprs.fi/#!z=11&amp;call=a%2FRufzeichen-SSID">Rufzeichen-SSID</a>

Wäre super, wenn das möglich wäre!

Vielen Dank

DL6ABE Andreas

elFallino, 2025-05-05:
Hallo Andreas, das wird machbar sein.
Ein paar Tage wird es ggf. dauern, die UI wird gerade überarbeitet.

elFallino, 2025-05-08:
Ich werde da die Notation "https://aprs.fi/?call=N0CALL" verwenden.
( siehe https://aprs.fi/page/linking )

Sollte das ungünstig sein, bitte gerne eine Rückmeldung :)

oe1kbc, 2025-05-18:
done v4.34x.05.18

closed 2025-05-18 by oe1kbc