icssw-org/MeshCom-Firmware #412: Missing Setup Options in v4.34x.05.18

mi-gri, 2025-05-19:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior. Please describe as clear as possible:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Serial Log**
If possible provide a serial log of the command --info as well if needed the output of the debug.
You can find the commands for enabling debugging on this page:
https://icssw.org/en/meshcom-kommandos-cl-gw/

If no serial Log is available, provide at least the firmware version you are running:

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - App Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

oe1kbc, 2025-05-19:
On the way

oe1kbc, 2025-05-19:
done v4.34x.05.19
- command line 
- OTA Update

oe1kbc, 2025-05-23:
done v4.,34y.05.23

closed 2025-05-23 by oe1kbc