icssw-org/MeshCom-Firmware #455: [BUG] possible Crash @ readGPS()

karamo, 2025-05-26:
**FW 4.34y**

In **esp32_gps.cpp** ist die Ablauflogik und Absicherung nicht "wasserdicht".

Der finale, minimalisiert Code:

```
unsigned int readGPS(void)
{
    #if defined(GPS_WAKEUP)
        pinMode(GPS_WAKEUP, OUTPUT);
        digitalWrite(GPS_WAKEUP, HIGH);
        delay(200);
    #endif

    if(bGPSDEBUG) Serial.println("-----------check GPS-----------");
  
    bool newData = false;
    unsigned long start = millis();

    while ((millis() - start) < 1000)
    {
        while (GPS.available())
        {
            char c = GPS.read();
            if(((c>=0x20) && (c<0x7f)) ||(c==0x0A)||(c==0x0D))
            {
                if (tinyGPSPlus.encode(c)) newData = true;
                if(bGPSDEBUG) Serial.print(c);
            }
        }
    }

    if(bGPSDEBUG)
        Serial.printf("newData:%i SAT:%d Fix:%d UPD:%d VAL:%d HDOP:%i\n", newData, tinyGPSPlus.satellites.value(), tinyGPSPlus.sentencesWithFix(), tinyGPSPlus.location.isUpdated(), tinyGPSPlus.location.isValid(), tinyGPSPlus.hdop.value());
...

```

karamo, 2025-05-26:
getestet mit **GPS M100-5883 mini @ 38400 Baud & 1 Hz Rate**

karamo, 2025-05-26:
ein gesamte Übertragungspaket von ≈500 Zeichen dauert:
@ 9600 Baud ≈ 500ms
@ 38400 Baud ≈ 130ms

Wenn aber auch periodisch in readGPS() 1s lang Zeichen empfangen werden, dann blockiert das besonders alle anderen Aktionen.
Es wäre daher ein modifiziertes Konzept über Interrupt anzuraten.

karamo, 2025-05-26:
**SoftwareSerial** würde zwar **115200 Baud** unterstützen, aber es zeigt sich, dass die Übertragung **nicht fehlerfrei** ist.
Es würde sich noch **57600 Baud als 3. Baud-Rate** anbieten.

oe1kbc, 2025-05-26:
done v4.34y.05.26