icssw-org/MeshCom-Firmware #490: [Feature] for ESP32-S3 N16R8 Boards 16MB partition

karamo, 2025-06-09:
**FW 4.34y.06.06**

in der allgemeinen **platformio.ini** wird unter **[esp32]**  nur eine **4M** partitiontable definiert:
`board_build.partitions = partitions-4MB-safeboot.csv`

Für die neuen **ESP32-S3-DevKitC-1-N16R8** Boards bitte eine **16MB Partition Table inkl. coredump partition** bereitstellen.

Es betrifft derzeit:
**E22_1262_S3-DevKitC-1-N16R8
E22_1268_S3-DevKitC-1-N16R8**

oe1kbc, 2025-06-09:
Hallo Wolfgang,
Rainer und ich haben es bereits mehrmals erklärt. So lange keine wirkliche Notwendigkeit besteht bleibt die Einteilung für alle Nodes gleich. Bitte das zu verstehen.

closed 2025-06-09 by oe1kbc

karamo, 2025-06-09:
Ok, war ja nur eine Erinnerung.
Wenn es noch nicht notwendig ist, dann kann es ja bleiben.