icssw-org/MeshCom-Firmware #534: [BUG] wrong Info

karamo, 2025-07-04:
**FW 4.35a**

In command_functions.cpp Z.3056 bei "wifitxpower" folgend stimmt die Aussage nicht:

[image]

hier gehört **"*"**, da der **Faktor = dBm * 4** berechnet wird.

[image]

karamo, 2025-07-04:
desweiteren ist in **udp_functions.cpp** der Faktor zu übergeben und nicht die Leistung in dBm, die ja im Flash gespeichert wird.
Daher muss auch hier der gespeicherte Wert *4 genommen werden, da die Konstante **WIFI_POWER_8_5dBm = 34** ist.

[image]

karamo, 2025-07-04:
siehe hier:
https://docs.espressif.com/projects/esp-techpedia/en/latest/esp-friends/advanced-development/performance/modify-tx-power.html#wi-fi-tx-power

oe1kbc, 2025-07-13:
v4.35a.07.13

closed 2025-07-13 by oe1kbc