icssw-org/MeshCom-Firmware #553: WLAN-Verbindung nach der MHeard-Abfrage getrennt.

HelmutEsterer, 2025-07-16:
Bei der Version 4.35b wird nach einer MHeard-Abfrage die WLAN-Verbindung getrennt und 
der Node muss neu gestartet werden.

HW: Heltec V3

vy 73 de

Helmut / DJ8MEH

karamo, 2025-07-16:
möglicherweise hängt es damit zusammen:

Hmm, irgendwas passiert doch ... nach einiger Zeit ... gesamte Log-Datei:

[V4-35b_lostWiFi.txt](https://github.com/user-attachments/files/21258513/V4-35b_lostWiFi.txt)

und hier passiert es:
```
13:38:30 [KEEP]...KEEP337F4EACOE3WAS-334.35b20;
13:39:00 [KEEP]...KEEP337F4EACOE3WAS-334.35b20;
13:39:29 [Web]...Client IP: [Web]...Client IP: 192.168.1.104
13:39:29 New Client.
13:39:29 GET /?page=mheard HTTP/1.1
13:39:29 Host: 192.168.1.176
13:39:29 Connection: keep-alive
13:39:29 User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/138.0.0.0 Safari/537.36
13:39:29 DNT: 1
13:39:29 Accept: */*
13:39:29 Referer: http://192.168.1.176/
13:39:29 Accept-Encoding: gzip, deflate
13:39:29 Accept-Language: de-DE,de;q=0.9,en-US;q=0.8,en;q=0.7
13:39:29 
13:39:29 Client disconnected.
13:39:29 
13:39:30 [KEEP]...KEEP337F4EACOE3WAS-334.35b20;
13:40:00 [KEEP]...KEEP337F4EACOE3WAS-334.35b20;
13:40:16 12:40:16 NEW-HEY: 029 @ xFD3AB25F H04 S0 T0 M00 OE3WAS-33>HG@R HW:48 MOD:8/8 FCS:06F8 FW:35:b LH:B0
13:40:16 12:40:16 Insert own_msg_id:FD3AB25F <FD3AB25F>
13:40:16 [DATA]...DATA337F4EACOE3WAS-334.35b   0   003
13:40:16 12:40:16 TX-UDP : 029 @ xFD3AB25F H04 S0 T0 M01 OE3WAS-33>HG@R HW:48 MOD:8/8 FCS:06F8 FW:35:b LH:B0
13:40:16 12:40:16 TX-LoRa: 029 @ xFD3AB25F H04 S0 T0 M01 OE3WAS-33>HG@R HW:48 MOD:8/8 FCS:06F8 FW:35:b LH:B0
13:40:16 [ERROR]...Sending UDP Packet failed
13:40:16 APRS decode - Packet discarded, wrong APRS-protocol - size <0> to short!
13:40:16 
13:40:16 12:40:16 TX-UDP : 000  x00000000 H02 S0 T0 M00 > HW:48 MOD:8/8 FCS:0000 FW:35:b LH:B0
13:40:16 [ERROR]...Sending UDP Packet failed
13:40:16 APRS decode - Packet discarded, wrong APRS-protocol - size <0> to short!
```

und eigenartig, was da passiert:
```
14:17:46 [Web]...Client IP: [Web]...Client IP: 192.168.1.104
14:17:46 New Client.
14:17:46 GET /?page=mheard HTTP/1.1
14:17:46 Host: 192.168.1.176
14:17:46 Connection: keep-alive
14:17:46 User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/138.0.0.0 Safari/537.36
14:17:46 DNT: 1
14:17:46 Accept: */*
14:17:46 Referer: http://192.168.1.176/
14:17:46 Accept-Encoding: gzip, deflate
14:17:46 Accept-Language: de-DE,de;q=0.9,en-US;q=0.8,en;q=0.7
14:17:46 
14:17:46 Client disconnected.
14:17:46 
14:17:46 
14:17:46 [WIFI]...connect OK
14:17:46 [WIFI]...now listening at IP 192.168.1.176, UDP port 1990
14:17:46 [WIFI]...Internet UDP-DEST meshcom.oevsv.at
14:17:47 [WIFI]...Internet NTP-DEST pool.ntp.org
14:17:47 [KEEP]...KEEP337F4EACOE3WAS-334.35b20;
14:17:47 [Web]...Client IP: [Web]...Client IP: 192.168.1.104
14:17:47 New Client.
14:17:47 GET /?page=mheard HTTP/1.1
14:17:47 Host: 192.168.1.176
14:17:47 Connection: keep-alive
14:17:47 User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/138.0.0.0 Safari/537.36
14:17:47 DNT: 1
14:17:47 Accept: */*
14:17:47 Referer: http://192.168.1.176/
[... 48 lines cut ...]
14:18:02 receiveflag
```

#### und dann ein CRASH:
[V4-35b_lostWiFi_CRASH.txt](https://github.com/user-attachments/files/21258556/V4-35b_lostWiFi_CRASH.txt)

#### Crash decodiert:
```
    0x420dedcd: _svfprintf_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1528
    0x420e6376: _vsnprintf_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdio/vsnprintf.c:70
    0x420e63b2: vsnprintf at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdio/vsnprintf.c:40
    0x42051da2: Print::printf(char const*, ...) at D:\...C:/Users/KARAHOAN/.platformio/packages/framework-arduinoespressif32/cores/esp32\Print.cpp:55

    0x420228e6: sub_page_rxlog() at D:\...src/web_functions\web_functions.cpp:719
    0x42025d9f: work_webpage(bool, int) at D:\...src/web_functions\web_functions.cpp:356
    0x4202614e: web_client_html(CommonWebClient) at D:\...src/web_functions\web_functions.cpp:218
    0x4202623d: loopWebserver() at D:\...src/web_functions\web_functions.cpp:134
    0x42013bdb: esp32loop() at D:\...src/esp32\esp32_main.cpp:2353
    0x4201c7db: loop() at D:\...src\main.cpp:65
    0x420569b9: loopTask(void*) at D:\...C:/Users/KARAHOAN/.platformio/packages/framework-arduinoespressif32/cores/esp32\main.cpp:50
```

In **web_functions.cpp**:719
`web_client.printf("<p class=\"font-small no-wrap\">%s</nobr></td></tr>\n", ringbufferRAWLoraRX[iRead]);`
da passiert der Crash💥

oe1kbc, 2025-07-16:
Danke für die Analyse, Fehler lag am neuen Ringbuffer-Handling

done 4.34c

closed 2025-07-16 by oe1kbc