icssw-org/MeshCom-Firmware #568: Gateway sendet nicht alle Pakete

HelmutEsterer, 2025-07-21:
Bis zum Gateway kommen alle Pakete, z. B. Userliste von db0sep-12, an.

- Damit scheiden Kollisionen "bis ins Haus" aus.
- Hingegen werden vom GW nicht alle Pakete an das Ziel gesendet.

Ich füge eine Dokumentation vom Gateway über Meshdash (dj8meh-8) und über den Web-Client
vom Ziel und Anfrager (dj8meh-46) an.

Hier scheint mir im Empfagnsbuffer oder Ringbuffer etwas durcheinander zu kommen.
- Oder es werden fehlende ACK-Meldungen nicht verarbeitet.

[image]
[image]

oe1kbc, 2025-07-21:
Bitte gib genau an welches packet Deiner Meinung nach nicht weiter gegeben wird. Bitte die 8-stelloge HEX MSG-ID angeben.
Danke

oe1kbc, 2025-07-21:
Ich lege das für einen späteren test zur seite

closed 2025-07-21 by oe1kbc